capn-freako/PyBERT #25: Errors and Warnings when run Python -m install

nyctalope, 2017-01-06:
I followed the instruction below using Anaconda2(64bit) 
**conda install enable
conda install traitsui
conda install chaco
conda install kiwisolver
conda install scipy
conda install Sphinx
conda install matplotlib
pip install PyBERT
python -m pybert**

Looks the installs are almost good except the last one when run "python -m pybert"
There are still several errors or warnings. Not sure if they are critical

C**:\python\Anaconda2\lib\site-packages\pybert-2.0.2-py2.7.egg\pybert\pybert.py:1139: VisibleDeprecationWarning: using a non-integer number instead of an integer will result in an error in the future
C:\python\Anaconda2\lib\site-packages\chaco\array_data_source.py:123: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._data = newdata
C:\python\Anaconda2\lib\site-packages\traits\trait_handlers.py:1599: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  if value in self.values:
C:\python\Anaconda2\lib\site-packages\chaco\image_data.py:190: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._data = newdata
Traceback (most recent call last):
  File "C:\python\Anaconda2\lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\python\Anaconda2\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "build\bdist.win-amd64\egg\pybert\__main__.py", line 3, in <module>
  File "C:\python\Anaconda2\lib\site-packages\traits\has_traits.py", line 2156, in configure_traits
    kind, handler, id, scrollable, args )
  File "C:\python\Anaconda2\lib\site-packages\traitsui\toolkit.py", line 245, in view_application
    raise NotImplementedError
NotImplementedError**

capn-freako, 2017-01-06:
Would you, please, try this:

*conda install wxPython*

and let me know if that works?

Thanks!
-db

capn-freako, 2017-01-06:
And would you also try this command, and send me the resultant output?

*conda list qt*

Thanks!
-db

nyctalope, 2017-01-06:
I just installed Anaconda2.0 and wxPython (wxPython is then added into the
package lib of Anaconda2.0) and have not install any packages, then I run
command "conda list qt",

*(**C:\python\Anaconda2) C:\Users\longyan\Documents>conda list qt*
*# packages in environment at C:\python\Anaconda2:*
*#*
*pyqt                      5.6.0                    py27_0*
*qt                        5.6.0                     vc9_0  [vc9]*
*qtawesome                 0.3.3                    py27_0*
*qtconsole                 4.2.1                    py27_2*
*qtpy                      1.1.2                    py27_0*

nyctalope, 2017-01-06:
My anaconda is Anaconda2-4.2.0-Windows-x86_64.exe
When run "conda install enable", looks some packages will be upgraded while
some downgraded.
Possibly QT has been downgraded....

*(C:\python\Anaconda2) C:\Users\longyan\Documents>conda install enable*
*Fetching package metadata .........*
*Solving package specifications: ..........*

*Package plan for installation in environment C:\python\Anaconda2:*

*The following packages will be downloaded:*

*    package                    |            build*
*    ---------------------------|-----------------*
*    conda-env-2.6.0            |                0          498 B*
*    numpy-1.10.4               |           py27_2         2.9 MB*
*    pyqt-4.10.4                |           py27_1        31.2 MB*
*    traits-4.5.0               |           py27_0         373 KB*
*    conda-4.2.13               |           py27_0         455 KB*
*    numexpr-2.6.0              |      np110py27_0         139 KB*
*    pyface-4.5.2               |           py27_0         736 KB*
*    scipy-0.17.1               |      np110py27_1        11.5 MB*
*    scikit-learn-0.17.1        |      np110py27_1         3.5 MB*
*    traitsui-4.5.1             |           py27_0         839 KB*
*    enable-4.5.1               |      np110py27_0         1.2 MB*
*    ------------------------------------------------------------*
*                                           Total:        52.9 MB*

*The following NEW packages will be INSTALLED:*

*    conda-env:    2.6.0-0*
*    enable:       4.5.1-np110py27_0*
*    pyface:       4.5.2-py27_0*
*    traits:       4.5.0-py27_0*
*    traitsui:     4.5.1-py27_0*

*The following packages will be UPDATED:*

*    conda:        4.2.9-py27_0       --> 4.2.13-py27_0*
*    scikit-learn: 0.17.1-np111py27_1 --> 0.17.1-np110py27_1*

*The following packages will be DOWNGRADED due to dependency conflicts:*

*    numexpr:      2.6.1-np111py27_0  --> 2.6.0-np110py27_0*
*    numpy:        1.11.1-py27_1      --> 1.10.4-py27_2*
*    pyqt:         5.6.0-py27_0       --> 4.10.4-py27_1*
*    scipy:        0.18.1-np111py27_0 --> 0.17.1-np110py27_1*

*Proceed ([y]/n)? y*

*Fetching packages ...*
*conda-env-2.6. 100% |###############################| Time: 0:00:00  83.00
kB/s*
*numpy-1.10.4-p 100% |###############################| Time: 0:00:01   2.43
MB/s*
*pyqt-4.10.4-py 100% |###############################| Time: 0:00:47 691.92
kB/s*
*traits-4.5.0-p 100% |###############################| Time: 0:00:00 561.96
kB/s*
*conda-4.2.13-p 100% |###############################| Time: 0:00:00 570.09
kB/s*
*numexpr-2.6.0- 100% |###############################| Time: 0:00:00 318.42
kB/s*
*pyface-4.5.2-p 100% |###############################| Time: 0:00:00 820.25
kB/s*
*scipy-0.17.1-n 100% |###############################| Time: 0:00:02   5.96
MB/s*
*scikit-learn-0 100% |###############################| Time: 0:00:01   2.45
MB/s*
*traitsui-4.5.1 100% |###############################| Time: 0:00:00 978.81
kB/s*
*enable-4.5.1-n 100% |###############################| Time: 0:00:00   1.31
MB/s*
*Extracting packages ...*
*[      COMPLETE      ]|##################################################|
100%*
*Unlinking packages ...*
*[pyqt                ]|#########################                         |
 50%*

nyctalope, 2017-01-06:
*I was thinking the problem is caused by package version mismatch, but not
so sure....*

*(C:\python\Anaconda2) C:\Users\longyan\Documents>conda list qt*
# packages in environment at C:\python\Anaconda2:
#
pyqt                      5.6.0                    py27_0
qt                        5.6.0                     vc9_0  [vc9]
qtawesome                 0.3.3                    py27_0
qtconsole                 4.2.1                    py27_2
qtpy                      1.1.2                    py27_0

*(C:\python\Anaconda2) C:\Users\longyan\Documents>conda install enable*
Fetching package metadata .........
Solving package specifications: ..........

Package plan for installation in environment C:\python\Anaconda2:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    conda-env-2.6.0            |                0          498 B
    numpy-1.10.4               |           py27_2         2.9 MB
    pyqt-4.10.4                |           py27_1        31.2 MB
    traits-4.5.0               |           py27_0         373 KB
    conda-4.2.13               |           py27_0         455 KB
    numexpr-2.6.0              |      np110py27_0         139 KB
    pyface-4.5.2               |           py27_0         736 KB
    scipy-0.17.1               |      np110py27_1        11.5 MB
    scikit-learn-0.17.1        |      np110py27_1         3.5 MB
    traitsui-4.5.1             |           py27_0         839 KB
    enable-4.5.1               |      np110py27_0         1.2 MB
    ------------------------------------------------------------
                                           Total:        52.9 MB

The following NEW packages will be INSTALLED:

    conda-env:    2.6.0-0
    enable:       4.5.1-np110py27_0
    pyface:       4.5.2-py27_0
    traits:       4.5.0-py27_0
    traitsui:     4.5.1-py27_0

The following packages will be UPDATED:

    conda:        4.2.9-py27_0       --> 4.2.13-py27_0
    scikit-learn: 0.17.1-np111py27_1 --> 0.17.1-np110py27_1

The following packages will be DOWNGRADED due to dependency conflicts:

    numexpr:      2.6.1-np111py27_0  --> 2.6.0-np110py27_0
    numpy:        1.11.1-py27_1      --> 1.10.4-py27_2
    pyqt:         5.6.0-py27_0       --> 4.10.4-py27_1
    scipy:        0.18.1-np111py27_0 --> 0.17.1-np110py27_1

Proceed ([y]/n)? y

Fetching packages ...
conda-env-2.6. 100% |###############################| Time: 0:00:00  83.00
kB/s
numpy-1.10.4-p 100% |###############################| Time: 0:00:01   2.43
MB/s
pyqt-4.10.4-py 100% |###############################| Time: 0:00:47 691.92
kB/s
traits-4.5.0-p 100% |###############################| Time: 0:00:00 561.96
kB/s
conda-4.2.13-p 100% |###############################| Time: 0:00:00 570.09
kB/s
numexpr-2.6.0- 100% |###############################| Time: 0:00:00 318.42
[... 307 lines cut ...]
"C:\python\Anaconda2\lib\site-packages\traitsui\editors\code_editor.py",
line 48, in ToolkitEditorFactory
    mark_color = Color( 0xECE9D8 )
  File "C:\python\Anaconda2\lib\site-packages\traits\traits.py", line 522,
in __call__
    return self.maker_function( *args, **metadata )
  File "C:\python\Anaconda2\lib\site-packages\traits\traits.py", line 1236,
in Color
    return ColorTrait( *args, **metadata )
  File "C:\python\Anaconda2\lib\site-packages\traitsui\toolkit_traits.py",
line 7, in ColorTrait
    return toolkit().color_trait( *args, **traits )
  File "C:\python\Anaconda2\lib\site-packages\traitsui\toolkit.py", line
161, in toolkit
    _toolkit = _import_toolkit(toolkit_name)
  File "C:\python\Anaconda2\lib\site-packages\traitsui\toolkit.py", line
83, in _import_toolkit
    return __import__( name, globals=globals(), level=1 ).toolkit
  File "C:\python\Anaconda2\lib\site-packages\traitsui\qt4\__init__.py",
line 18, in <module>
    import pyface.qt
  File "C:\python\Anaconda2\lib\site-packages\pyface\qt\__init__.py", line
44, in <module>
    % qt_api)
RuntimeError: Invalid Qt API 'pyqt5', valid values are: 'pyqt' or 'pyside'

capn-freako, 2017-01-06:
> Hi David
> Finally I got anything done!
>
> 1. Download the Anaconda2 from website
> Anaconda2-4.2.0-Windows-x86_64.exe
>
> 2. I was not using the command line, instead I tried to upgrade or install
> these package using Anaconda Navigator, shown as below
> a) select all, that will include both installed or uninstalled package,
> and type the package name to search.
> b) you may find some package names there but not installed, click "Apply"
> to install.
> [image: Inline image 1]
>
> 3. Install PyBERT by clicking Root -> Open Terminal, and run the command
> "pip install PyBERT" in the shell.
> [image: Inline image 2]
>
> [image: Inline image 3]
>
> 4. Launch PyBERT by command "python -m pybert" in the shell.
> [image: Inline image 5]
>
> [image: Inline image 6]
>
> done!
>
> Excited!
>
> thanks for your great design!!
> -Long

Okay, glad to hear you’re up and running!

Thanks for all the feedback,
-db

closed 2017-05-05 by capn-freako