capn-freako/PyBERT #26: trouble with installation

jmecherul, 2017-01-23:
Hi

I am trying to follow your guide in Win10, but I am getting the following error:

C:\Users\cmuntean\Desktop\PyBERT-2.0.5>pip install PyBert
Collecting PyBert
  Using cached PyBERT-2.0.5.tar.gz
Collecting traits (from PyBert)
  Using cached traits-4.6.0.zip
Collecting traitsui (from PyBert)
  Using cached traitsui-5.1.0.zip
Collecting enable (from PyBert)
  Using cached enable-4.6.1.tar.bz2
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "c:\users\cmuntean\appdata\local\temp\pip-build-zgqdlz\enable\setup.py", line 35, in <module>
        from numpy.distutils.core import setup
    ImportError: No module named numpy.distutils.core

    ----------------------------------------

I am no sure what to do.   You help is greatly appreciated.  Thank you

Catalin

capn-freako, 2017-01-23:
Hi Catalin,

Thanks for your interest in PyBERT, and I’m sorry you’re having trouble.

It looks to me like you don’t have NumPy installed, but let’s do a few sanity checks.
Please, execute the following commands and send me the resultant output for each:

which python
echo PYTHON_PATH
python -c "import sys; print(sys.path)”
python -c "import numpy; print(numpy.__file__)”

Note that, in #4, there are two underscores before and after “file”.

Thanks,
-db

> On Jan 23, 2017, at 5:52 AM, jmecherul <notifications@github.com> wrote:
> 
> numpy.distutils.core

jmecherul, 2017-01-23:
Hi

Thank you.  Here are the outputs.

(C:\Program Files\Anaconda2) C:\WINDOWS\system32>which python
'which' is not recognized as an internal or external command,
operable program or batch file.

(C:\Program Files\Anaconda2) C:\WINDOWS\system32>echo PYTHON_PATH
PYTHON_PATH

(C:\Program Files\Anaconda2) C:\WINDOWS\system32>python -c "import sys; print(sys.path)"
['', 'C:\\Program Files\\Anaconda2\\python27.zip', 'C:\\Program Files\\Anaconda2\\DLLs', 'C:\\Program Files\\Anaconda2\\lib', 'C:\\Program Files\\Anaconda2\\lib\\plat-win', 'C:\\Program Files\\Anaconda2\\lib\\lib-tk', 'C:\\Program Files\\Anaconda2', 'C:\\Program Files\\Anaconda2\\lib\\site-packages', 'C:\\Program Files\\Anaconda2\\lib\\site-packages\\PIL', 'C:\\Program Files\\Anaconda2\\lib\\site-packages\\Sphinx-1.5.1-py2.7.egg', 'C:\\Program Files\\Anaconda2\\lib\\site-packages\\win32', 'C:\\Program Files\\Anaconda2\\lib\\site-packages\\win32\\lib', 'C:\\Program Files\\Anaconda2\\lib\\site-packages\\Pythonwin', 'C:\\Program Files\\Anaconda2\\lib\\site-packages\\setuptools-27.2.0-py2.7.egg', 'C:\\Program Files\\Anaconda2\\lib\\site-packages\\wx-3.0-msw']

(C:\Program Files\Anaconda2) C:\WINDOWS\system32>python -c "import numpy: pint(numpy.__file__)"
  File "<string>", line 1
    import numpy: pint(numpy.__file__)
                ^
SyntaxError: invalid syntax

(C:\Program Files\Anaconda2) C:\WINDOWS\system32>python -c "import numpy; pint(numpy.__file__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'pint' is not defined

(C:\Program Files\Anaconda2) C:\WINDOWS\system32>python -c "import numpy; print(numpy.__file__)"
C:\Program Files\Anaconda2\lib\site-packages\numpy\__init__.pyc

capn-freako, 2017-01-23:
> File "", line 1
> File "", line 1, in

Okay, please, try this:

conda install enable

-db

jmecherul, 2017-01-23:
I did.  I ran all commands in the "instant gratification" tap in the wiki.  

enable 4.5.1 already installed.  

played a little with things, and it seem that installing scipy after the chaco will break things up

(C:\Program Files\Anaconda2) C:\WINDOWS\system32>conda install scipy
Fetching package metadata ...........
Solving package specifications: .

Package plan for installation in environment C:\Program Files\Anaconda2:

The following packages will be UPDATED:

    numexpr:      2.4.6-np110py27_0  --> 2.6.1-np111py27_2
    numpy:        1.10.1-py27_0      --> 1.11.3-py27_0
    scikit-learn: 0.17-np110py27_1   --> 0.18.1-np111py27_1
    scipy:        0.16.0-np110py27_0 --> 0.18.1-np111py27_1

Proceed ([y]/n)? y

Did this and ran python -m pybert  and got this

(C:\Program Files\Anaconda2) C:\WINDOWS\system32>python -m pybert
C:\Program Files\Anaconda2\lib\site-packages\pybert\pybert.py:1146: VisibleDeprecationWarning: using a non-integer number instead of an integer will result in an error in the future
  p[clock_pos + nspui * (0.5 + i) :] -= p[clock_pos + nspui * (1 + i)]
C:\Program Files\Anaconda2\lib\site-packages\chaco\array_data_source.py:123: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._data = newdata
C:\Program Files\Anaconda2\lib\site-packages\traits\trait_handlers.py:1599: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  if value in self.values:
C:\Program Files\Anaconda2\lib\site-packages\chaco\image_data.py:190: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  self._data = newdata
swig/python detected a memory leak of type 'wxPyHtmlWinTagHandler *', no destructor found.
swig/python detected a memory leak of type 'wxPyHtmlWinTagHandler *', no destructor found.
swig/python detected a memory leak of type 'wxPyHtmlWinTagHandler *', no destructor found.

script started by gave these above errors.

and in the program 
10:18:00: iCCP: known incorrect sRGB profile
10:18:00: iCCP: known incorrect sRGB profile

jmecherul, 2017-01-23:
when closing the script i get:

..\..\src\common\module.cpp(194): assert "module -> m_state == State_initialized" failed in wxModule::DoCleanUpModules(): not initialized module being cleaned up

capn-freako, 2017-05-05:
Hi Catalin,

Are you still struggling with this?

Thanks,
-db

capn-freako, 2017-05-09:
I haven't heard from Catalin, since my last comment. So, I'm closing this.

One final note: I notice several "wx..." entries in the log, above. I suspect this may mean that Catalan's system was attempting to use the _wx_, as opposed to the _qt4_, backend. This is known to be problematic. I now force the choice with the following code, in _pybert.py_:

```
from traits.trait_base import ETSConfig
ETSConfig.toolkit = 'qt4'
```

This small change may make this problem go away.
Catalin, if you happen to return and decide to take the time to try the current version, please, let me know if this change fixed your issue. Thanks!

closed 2017-05-09 by capn-freako