capn-freako/PyBERT #30: Freq. Resp. of channel from file fails to plot

JEprime, 2017-11-14:
Hi, 

When I put an impulse response into PyBERT from a csv file, it fails to plot the frequency response. However, all other outputs look OK.

Jason
[image]

capn-freako, 2017-12-10:
Hi Jason,

Thanks for reporting this issue!

I've played with this quite a bit, using many different channels I have available, and have concluded that there's something unique about the channel you're using, which is triggering this behavior. Are you able to post the *.CSV file you're using?

Thanks,
-db

JEprime, 2017-12-11:
Hi David,

Yes, I can share them. Should I post them or send them directly?

Cheers,
Jason

capn-freako, 2017-12-11:
Either way is fine.

Thanks!
-db

JEprime, 2017-12-11:
Here you go!

Jason

capn-freako, 2017-12-11:
I don’t see any attached file(s).
-db

JEprime, 2017-12-11:
> auth/AA_

Does your email address accept csv files?

capn-freako, 2017-12-12:
Yes, as far as I know.

capn-freako, 2017-12-13:
Hi Jason,

I went back through the e-mail thread and found this file attached to a message you sent me on Nov. 13: _7m30awg.csv_. Is that the correct file? If so, I think the problem may be the magnitudes of the numbers in the second column. I'm assuming this is an impulse response, in which case its second column should have units **(V/s)**. I don't think there's any way these numbers could have those units. Here's a plot of that file:

[image]

JEprime, 2017-12-13:
Hi David,

Yes, that's a good file.

I obtained this impulse response from S-parameters by extrapolating a dc
point, enforcing symmetry, and doing an ifft.

The impulse response is unitless because it is a time domain transfer
function.

I'm not sure how exactly to convert this to volts/s. Do you have a method?

Thank you,
Jason

<https://user-images.githubusercontent.com/1047450/33950142-75e29c7a-dfe0-11e7-8a49-28e7fd163f7c.png>

capn-freako, 2017-12-13:
Just normalize it to the sample interval of your signal vector.

JEprime, 2017-12-13:
In other words, divide by delta-t?

capn-freako, 2017-12-13:
Yes.

commit 8c63d282 (2017-12-14): Rel. v2.3.1

Fixes #30
Fixes #36
Fixes #37
closed 2017-12-14 by capn-freako