capn-freako/PyBERT #31: Add COM metric reporting.

capn-freako, 2017-12-03:
The Channel Operating Margin (COM) method of measuring high speed serial communication channel performance is becoming more and more popular. I'd like to add this ability to PyBERT.

capn-freako, 2023-12-25:
I think [this Anritsu white paper](https://dl.cdn-anritsu.com/en-us/test-measurement/files/Technical-Notes/White-Paper/11410-00989A.pdf) is a great reference for the details of COM calculation.
(In particular, see Sec. 3.)

Most other references I'm finding tend to refer the reader to M-code scripts and MS _Excel_ spreadsheets, which aren't very helpful to us writing Python code.

dracid, 2024-02-25:
I've been hunting the COM code for some time :))
It's actually openly accessible here:
https://www.ieee802.org/3/df/public/tools/index.html
[mellitz_3dj_02_elec_230817_COM_v4p1_update.zip](https://github.com/capn-freako/PyBERT/files/14395711/mellitz_3dj_02_elec_230817_COM_v4p1_update.zip)
[mellitz_3df_02_2211_com_v3p9.zip](https://github.com/capn-freako/PyBERT/files/14395712/mellitz_3df_02_2211_com_v3p9.zip)

version 4.1 has the latest code, but no config sheets.
version 3.9 has the code AND a few example of the config sheets.

Yes, it's an M file for Matlab (not Octave). I haven't yet found if anyone has re-implemented it in python.

capn-freako, 2024-02-25:
> Yes, it's an M file for Matlab (not Octave). I haven't yet found if anyone has re-implemented it in python.

Thanks, @dracid !
I've taken a first stab at this and am wondering if you might have time to look over my code and comment:
https://github.com/capn-freako/PyBERT/blob/860e34a068084d1865857b2ee3e48c2a871cd717/src/pybert/utility.py#L1487

Thanks!
-db