capn-freako/PyBERT #40: Convert code to be Python 3 compliant.

capn-freako, 2018-02-02:
More and more people are requesting Python 3 compatibility.
Make it so.

capn-freako, 2018-09-16:
Hm, this may be on hold until Anaconda brings in *Chaco v4.6*:
https://stackoverflow.com/questions/44586253/using-chaco-in-python-3-6-anaconda

capn-freako, 2018-09-16:
Read this link, if the above warning didn't dissuade you:
https://github.com/enthought/chaco/issues/319

capn-freako, 2018-09-16:
This, also, may or may not be pertinent:
https://github.com/enthought/enable/pull/242

capn-freako, 2018-09-16:
Here's a question I asked on SO about 18 months ago, which pertains:
https://stackoverflow.com/questions/44346018/conda-build-ignoring-my-activated-environment

jdpatt, 2019-02-05:
I've done some work to make it compatible with both 2/3 so that PyBERT can be used with python 3. You should run tox from the base conda environment. Running from another environment causes issues when generating virtualenvs.

Using the env under the instant gratification wiki and installing my branch; I've confirmed that PyBERT still works (python2.7).  Right now my biggest issue is getting enable to install from pip in 2 or 3.  https://github.com/enthought/enable/issues/329  Until that works; tox won't be able to build clean test environments and conda doesn't have a 3+ version.

https://github.com/jdpatt/PyBERT

capn-freako, 2019-03-01:
Hi David,

I just posted [this Quora question](https://www.quora.com/unanswered/How-can-I-make-my-enable-dependent-project-Python3-friendly), re: making `enable`-dependent projects **Python3** friendly.

jdpatt, 2019-03-02:
Awesome I got enable installing in Linux, so I'll figure out x11 forwarding in Docker or install a virtual machine to keep some momentum going.  Maybe Quora or the other issue will pull through for OS X!

capn-freako, 2019-09-18:
Thank you for driving this, David!

closed 2019-09-18 by capn-freako