capn-freako/PyBERT #47: RuntimeError: Invalid Qt API 'pyqt5', valid values are: 'pyqt' or 'pyside'

denzchoe, 2019-04-03:
Trouble running PyBert on my Win64 machine lately. I tried it a month back didn't have any issue. Not sure what change. I tried the instructions from https://github.com/capn-freako/PyBERT/wiki/instant_gratification

Output log:

`(pybert64) C:\Users\denzchoe>pybert
Traceback (most recent call last):
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\Scripts\pybert
-script.py", line 5, in <module>
    from pybert.pybert import main
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\lib\site-packa
ges\pybert\pybert.py", line 38, in <module>
    from pyface.api      import FileDialog, OK
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\lib\site-packa
ges\pyface\api.py", line 16, in <module>
    from about_dialog import AboutDialog
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\lib\site-packa
ges\pyface\about_dialog.py", line 18, in <module>
    from toolkit import toolkit_object
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\lib\site-packa
ges\pyface\toolkit.py", line 77, in <module>
    _init_toolkit()
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\lib\site-packa
ges\pyface\toolkit.py", line 38, in _init_toolkit
    be = import_toolkit(ETSConfig.toolkit)
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\lib\site-packa
ges\pyface\toolkit.py", line 31, in import_toolkit
    __import__(be + 'init')
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\lib\site-packa
ges\pyface\ui\qt4\init.py", line 18, in <module>
    from pyface.qt import QtCore, QtGui, qt_api
  File "C:\Users\denzchoe\AppData\Local\conda\conda\envs\pybert64\lib\site-packa
ges\pyface\qt\__init__.py", line 39, in <module>
    % qt_api)
RuntimeError: Invalid Qt API 'pyqt5', valid values are: 'pyqt' or 'pyside'`

But I found a temporary fix for now;
[hyperlink](https://github.com/acoular/acoular/issues/11#issuecomment-281435000)

by inserting a few lines to pybert-script.py
`import os`
`os.environ["QT_API"] = "pyqt"`

capn-freako, 2019-09-18:
Please, try reinstalling, using the _**new and recently changed**_ quick start instructions:
https://github.com/capn-freako/PyBERT/wiki/instant_gratification
and let me know, if you still encounter trouble.
Thanks!

capn-freako, 2019-10-24:
Closing after no response from OP.

closed 2019-10-24 by capn-freako