capn-freako/PyBERT #49: Can't install pybert package.

haykding, 2019-08-06:
I've created pybert64 env and did this: conda install -c dbanas pybert 

Anaconda gave the following errors... 

Collecting package metadata (current_repodata.json): done
Solving environment: failed with current_repodata.json, will retry with next rep
odata source.
Initial quick solve with frozen env failed.  Unfreezing env and trying again.
Solving environment: failed with current_repodata.json, will retry with next rep
odata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current cha
nnels:

How can I install it manually ? Seems anaconda can't find pybert package in it's cloud.

capn-freako, 2019-08-06:
Try this command:

```
conda install -c dbanas -c defaults -c conda-forge pybert
```

-db

haykding, 2019-08-07:
Nearly the same is happening ...
(pybert64) C:\Users\Administrator>conda install -c dbanas -c defaults -c conda-f
orge pybert
Collecting package metadata (current_repodata.json): done
Solving environment: failed with current_repodata.json, will retry with next rep
odata source.
Initial quick solve with frozen env failed.  Unfreezing env and trying again.
Solving environment: failed with current_repodata.json, will retry with next rep
odata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current cha
nnels:

  - chaco

Current channels:

  - https://conda.anaconda.org/dbanas/win-64
  - https://conda.anaconda.org/dbanas/noarch
  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch
  - https://conda.anaconda.org/conda-forge/win-64
  - https://conda.anaconda.org/conda-forge/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

haykding, 2019-08-07:
But I see pybert package in the repo
[image]

haykding, 2019-08-07:
Also I've added the anaconda channel for the chaco pkg. ,
`conda install -c dbanas -c defaults -c conda-forge -c anaconda pybert` 

but now it gives me 

> UnsatisfiableError: The following specifications were found to be incompatible with each other

capn-freako, 2019-08-07:
I’m sorry, I just realized you’re on a Windows machine.
This is currently not working for Windows, only MacOS X.
We’re trying to get it working for Windows and will post when we have done so.

-db

haykding, 2019-08-07:
Thank you ,  Can I get it work in the linux instead ?

capn-freako, 2019-08-07:
Not yet.

capn-freako, 2019-09-18:
Okay, the quick start instructions should work on all 3 platforms now.
Please, note that those instructions have changed: https://github.com/capn-freako/PyBERT/wiki/instant_gratification
Let me know, if things still aren't working for you.

haykding, 2019-09-29:
Thanks you very much !!! It's working now !

closed 2019-09-29 by haykding