capn-freako/PyBERT #52: Cannot load S-parameter file

dghan0219, 2019-09-15:
Using Anaconda3 2019.07 (64-bit) and PyBERT 3.1.1, opening a channel description file (S-parameters, pulse, or impulse) produces a save window rather than an open window.  Hence, the file cannot be loaded.

dghan0219, 2019-09-15:
...Hence the file cannot be loaded with the GUI.  Typing the full path to the file works and PyBERT runs with it, but relative paths to the file do not work and PyBERT cannot find it, although it can be hard to tell in which folder PyBERT thinks it is and that may be the problem.

jdpatt, 2019-09-18:
The traits module is still weird to me.  This also affects the tx and rx's ami/dll file open dialogs.  They say "save" but I still think they correctly return a file path.  

A lot of paths were originally done using os.path, we should continue to move to pathlib so that it doesn't matter relative or full.

dghan0219, 2019-09-18:
If I pick a channel description file and then click Save, I then get an overwrite confirmation, so it is not operating correctly in that case.  I have not tried the other dialogs except for the configuration buttons.

PR #57 (2019-09-28): Override the editors for channel, tx/rx ami and tx/rx dll.
Changes all the save menus to open menus and a minor spelling fix.

Should resolve #52
closed 2019-09-28 by capn-freako