capn-freako/PyBERT #56: Hover Help Not Working in PyBERT 3.1.1

dghan0219, 2019-09-24:
Hover help not working in PyBERT 3.1.1.  I'm using Anaconda3 2019.07.

jdpatt, 2019-09-28:
All hover tooltips or just one or two?  System/Version?

OSX 10.14 seems to work without issue.

[image]

dghan0219, 2019-09-28:
I'm using Windows 10 and none of the tooltips work on any tab.

dghan0219, 2019-09-28:
Windows 7 has the same problem.

jdpatt, 2019-09-29:
Seems to work on my W10 machine.

> conda list anaconda$
anaconda                  2019.07                  py37_0

dghan0219, 2019-09-29:
I see it's working now for all three Windows 10 PCs that I have and the one Windows 7 PC.  I think I was hovering over the text description of the input box rather than the input box.
This issue can be closed.

closed 2019-09-29 by dghan0219