capn-freako/PyBERT #65: Add issue/feature request templates.

jdpatt, 2019-10-27:
We should look into adding these templates!

OS, Python Version and PyBERT version being the ones of interest under the version section.

[Github Issue Templates](https://help.github.com/en/github/building-a-strong-community/creating-issue-templates-for-your-repository)

capn-freako, 2019-12-08:
I've gone through the preliminary instructions at the link you provided, but am not sure how to incorporate the values you mention (OS, Python version, and PyBERT version).

jdpatt, 2019-12-08:
It would be changing:

```
**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]
```
to

```
**Desktop (please complete the following information):**
 - OS: [e.g. Windows, OS X]
 - Python Version [e.g. 3.7.4]
 - PyBERT Version [e.g. 3.0.0]
```

capn-freako, 2019-12-08:
Thanks. Done.

closed 2019-12-08 by capn-freako