capn-freako/PyBERT #69: IBIS-AMI Issues

dghan0219, 2019-11-08:
Same issue as #52, but with the IBIS-AMI file dialogs this time -- saving rather than opening.

Also, why are all the check boxes associated with the IBIS-AMI models inactive?

For the Config buttons, please display the help text that is in the IBIS-AMI file next to the option.  Also, it appears only adaptive parameters are configurable and not the manually adjusted parameters as those boxes allow no changes.

capn-freako, 2019-12-08:
Hi Dennis,

Would you, please, pull the tx-rx-split branch from the main PyBERT GitHub repository and try running that version.
See if this issue is still pertinent when running that code, and let me know if so.

Thanks!
-db

capn-freako, 2019-12-28:
Fixed in v3.3.0.

closed 2019-12-28 by capn-freako