capn-freako/PyBERT #79: PyBERT 3.3.0 IBIS-AMI problems

dghan0219, 2020-01-04:
Most AMI options are being presented, but modulation is not one of them.  Most IBIS-AMI files have PAM4 set as the default, so the only way to change the modulation to NRZ for PyBERT's use is to edit the AMI file with a text editor.  Can this be handled by reading and passing the modulation type from the Config. tab?

All IBIS-AMI models I've tried create a simulation exception: TypeError: can't multiply sequence by non-int of type 'float'.

PyBERT cannot find the AMI information in most IBIS models that have S-parameter package models integrated into the model (integration was done by HyperLynx VX.2.6 and the integrated models work with HyperLynx VX.2.6 and ADS 2020).  It finds the AMI information in the equivalent IBIS model that doesn't have the package model integrated into it.

capn-freako, 2020-01-05:
Dennis,

1. I'm looking into the modulation issue.

1. Can you point me to a model that gives this error?

1. Can you send me such an "integrated" model to play with?

Thanks!

capn-freako, 2022-09-26:
Hi Dennis,

Is this still an issue?

Thanks,
-db

closed 2022-10-07 by capn-freako