capn-freako/PyBERT #84: IBIS model imports dumps info to console, even w/ debugging mode disabled.

capn-freako, 2020-02-27:
**Describe the bug**
Even when I uncheck the `Debug` flag on the `Config`. tab, PyBERT continues to dump copious amounts of debugging information to the console when I import an IBIS model.

**To Reproduce**
Steps to reproduce the behavior:
1. Uncheck the `Debug` flag on the `Config`. tab.
2. Import an IBIS model.
3. Observe the command prompt window from which you launched PyBERT.
4. Note the voluminous debugging information being printed out.

**Expected behavior**
With the `Debug` flag disabled, this information should _not_ be printed to the console.

**Screenshots**
(n/a)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Python Version 3.7.6
 - PyBERT Version 3.3.2

**Additional context**
(n/a)

jdpatt, 2021-12-07:
@capn-freako , you've got yourself assigned to this one.  Have you started work on it? Otherwise, I have a decent understanding what is happening and will reassign it plus #90 for the last of this little burst of effort.

capn-freako, 2021-12-08:
@jdpatt , this is fixed in my local sandbox.
So, you can skip it.

Thanks!

commit 5d2967d1 (2021-12-10): Combined v3.5.0.0 release.

Fixes #100
Fixes #97
Fixes #96
Fixes #95
Fixes #84
Fixes #77
closed 2021-12-10 by capn-freako