capn-freako/PyBERT #89: Command-line Options

dghan0219, 2020-05-14:
Add command-line options for PyBERT when it loads to do one, multiple, or all of the following...
1. Load configuration
2. Save configuration
3. Load results
4. Save results
5. Run
6. Exit with options to save results and/or configuration or no option at all
7. Repeat with another set of options

A possible scenario is load config, run, save results, exit.  Another scenario might be to load config, load results, run, and then pass to manual control.

capn-freako, 2021-12-09:
This sounds like something better served, via a simple tool scripting language, ala Tcl.
I think that might be a more appropriate and extensible solution than command line options.
What do you think?

dghan0219, 2021-12-09:
I've only used Tcl for very basic stuff, so maybe I don't understand enough about it.  It seems to me command-line options for PyBERT would still be needed for the Tcl script.

capn-freako, 2022-09-26:
**Attention helpers!** Please, use the branch created for this issue: `89-command-line-options`.

Thanks!
-db

unbtorsten, 2022-10-03:
Why not use something like argparse, cf. https://docs.python.org/3/library/argparse.html?