capn-freako/PyBERT #93: Save & Load Configuration Not Working in Version 4.4.1?

denzchoe, 2021-09-16:
running Python version 3.8.11

As per the title, saving and loading configuration doesn't work.
File is created during the save attempt but it is an empty file.

[image]

The following error log was reported:

```
Traceback (most recent call last):
  File "C:\Users\Denz\anaconda3\envs\PYBERT_Env\lib\site-packages\pybert\pybert_view.py", line 78, in do_save_cfg
    pickle.dump(the_PyBertCfg, the_file)
TypeError: write() argument must be str, not bytes

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Denz\anaconda3\envs\PYBERT_Env\lib\site-packages\traitsui\qt4\ui_base.py", line 57, in perform
    self.ui.do_undoable(handler.perform, self.ui.info, self.action, None)
  File "C:\Users\Denz\anaconda3\envs\PYBERT_Env\lib\site-packages\traitsui\ui.py", line 655, in do_undoable
    action(*args, **kw)
  File "C:\Users\Denz\anaconda3\envs\PYBERT_Env\lib\site-packages\traitsui\handler.py", line 217, in perform
    method(info)
  File "C:\Users\Denz\anaconda3\envs\PYBERT_Env\lib\site-packages\pybert\pybert_view.py", line 83, in do_save_cfg
    the_pybert.log("Exception raised by pybert.pybert_view.MyHandler.do_save_cfg().",
  File "C:\Users\Denz\anaconda3\envs\PYBERT_Env\lib\site-packages\pybert\pybert.py", line 608, in log
    raise exception
RuntimeError: The following error occured:
        {err}
The configuration was NOT saved.
```

capn-freako, 2021-09-16:
Try changing line 75 of pybert_view.py from this:

                with open(dlg.path, "w") as the_file:

to this:

                with open(dlg.path, "wb") as the_file:

(“w” ==> “wb”).

-db

denzchoe, 2021-09-17:
Found the fix.
According to documentation here;
https://github.com/enthought/pyface/blob/9ad6e3a73dae6a52289f8b1d04016451ac325192/pyface/image/image.py#L145

The line from here should be edited from;
https://github.com/capn-freako/PyBERT/blob/f9678eaa2a09491fbff9b684cfbde91e42b7eb04/pybert/pybert_view.py#L93

to 

                `with open(dlg.path, "rb") as the_file:`

If anyone else can verify this, I guess the issue can be closed?

commit 5d97d14f (2021-10-03): Fixed issues raised by @denzchoe.

Fixes #92
Fixes #93
Fixes #94
closed 2021-10-03 by capn-freako