capn-freako/PyBERT #101: PyBERT 3.5.1 simulations not finishing

dghan0219, 2021-12-14:
Simulations with PyBERT 3.5.1 using Anaconda3 2021.11 that has Python 3.8.12 (does PyBERT force this version rather than using 3.9?) fail using a channel file and using native channel definition.

With 4-port S-parameter file (Touchstone 1.0 and 2.0 formats), error is:
   ts2N = import_freq(filename)
TypeError: import_freq() missing 1 required positional argument: 'freqs'

With native channel definition, error occurs during plotting:
 start_ix = where(clock_times > start_t)[0][0]
TypeError: '>' not supported between instances of 'list' and 'int'

dghan0219, 2021-12-14:
I got the native channel definition to simulate by using the default values for Nbits, Nspb, and EyeBits.

capn-freako, 2021-12-14:
I've confirmed that a `*.s4p` channel file crashes, as Dennis reports:

```
(pybert)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (master)
$ pybert
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\capnf\AppData\Local\Continuum\anaconda3\envs\pybert\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\capnf\AppData\Local\Continuum\anaconda3\envs\pybert\lib\site-packages\pybert\pybert_view.py", line 46, in run
    my_run_sweeps(self.the_pybert)
  File "C:\Users\capnf\AppData\Local\Continuum\anaconda3\envs\pybert\lib\site-packages\pybert\pybert_cntrl.py", line 104, in my_run_sweeps
    my_run_simulation(self)
  File "C:\Users\capnf\AppData\Local\Continuum\anaconda3\envs\pybert\lib\site-packages\pybert\pybert_cntrl.py", line 191, in my_run_simulation
    chnl_h = self.calc_chnl_h()
  File "C:\Users\capnf\AppData\Local\Continuum\anaconda3\envs\pybert\lib\site-packages\pybert\pybert.py", line 1561, in calc_chnl_h
    ch_s2p_pre = import_channel(self.ch_file, ts, self.f)
  File "C:\Users\capnf\AppData\Local\Continuum\anaconda3\envs\pybert\lib\site-packages\pybert\pybert_util.py", line 854, in import_channel
    ts2N = import_freq(filename)
TypeError: import_freq() missing 1 required positional argument: 'freqs'
```

capn-freako, 2021-12-14:
@dghan0219 , can you tell me what combination of: `Nbits`, `Nspb`, and `EyeBits`, causes the native channel case to crash?

dghan0219, 2021-12-14:
eye_bits: 4096
nbits: 8192
nspb: 256

capn-freako, 2021-12-14:
That configuration works for me.

capn-freako, 2021-12-14:
@dghan0219 , would you save the configuration for the native channel failing case and send it to me, please?

capn-freako, 2021-12-14:
Okay, @dghan0219 , I just posted v3.5.2.
Please, update and give it a try.

dghan0219, 2021-12-14:
Here is my configuration file:

!!python/object:pybert.pybert_cfg.PyBertCfg
R0: 3.04788976566444
Rdc: 3.55650097068021
Theta0: 0.03448151
Z0: 102.3
alpha: 0.01
bit_rate: 9.8304
cac: 0.1
ch_file: JESD204B_9.8304Gbps_SL_gnuplot_s4p_mm.s4p
cin: 0.01
cout: 0.24938159035357
ctle_file: ''
ctle_mode: Manual
ctle_mode_tune: AGC
ctle_offset: 0.0
ctle_offset_tune: 0.0
debug: false
decision_scaler: 0.5
delta_t: 0.1
do_sweep: false
eye_bits: 1600
f_step: 5.2568049198524
gain: 0.5
impulse_length: 0.0
l_ch: 0.2286
lock_sustain: 500
mod_type:
- 0
n_ave: 100.0
n_lock_ave: 500
n_taps: 16
nbits: 8000
nspb: 32
num_sweeps: 1
pattern_len: 127
peak_freq: 4.9152
peak_mag: 7.37350445002135
pn_freq: 0.437
pn_mag: 0.001
rel_lock_tol: 0.1
rin: 100.0
rn: 0.001
rs: 100.0
rx_ami_file: ''
rx_bw: 49.152
rx_dll_file: ''
rx_ibis_file: ''
rx_use_ami: false
rx_use_getwave: false
rx_use_ibis: false
rx_use_ts4: false
sum_bw: 9.8304
sum_ideal: false
sweep_aves: 1
sweep_num: 1
thresh: 6
tx_ami_file: ''
tx_dll_file: ''
tx_ibis_file: ''
tx_tap_tuners:
- !!python/tuple
  - false
  - 0.0
- !!python/tuple
  - false
  - 0.0
- !!python/tuple
  - false
  - 0.0
- !!python/tuple
  - false
  - 0.0
tx_taps:
- !!python/tuple
  - false
  - 0.0
- !!python/tuple
  - false
  - 0.0
- !!python/tuple
  - false
  - 0.0
- !!python/tuple
  - false
  - 0.0
tx_use_ami: false
tx_use_getwave: false
tx_use_ibis: false
tx_use_ts4: false
use_ch_file: false
use_ctle_file: false
use_dfe: false
use_dfe_tune: false
v0: 0.497656617333547
vod: 0.2825
w0: 274172618.683821

dghan0219, 2021-12-14:
Version 3.5.2 works as intended with S-parameters and the native channel with my values of eye_bits, nbits, nspb and the default values.

capn-freako, 2021-12-14:
Great!
Thanks so much for your continued debugging support.
It really is invaluable. :)

Happy holidays,
-db

Fixed in 3.5.2

closed 2021-12-14 by capn-freako