capn-freako/PyBERT #104: Configuration file not being read correctly

dghan0219, 2022-07-21:
I have attached a configuration file I had PyBERT 3.5.5 load and then I saved it with a new name, also attached.  PyBERT is not reading some fields and is using defaults instead.
[PyBERT.zip](https://github.com/capn-freako/PyBERT/files/9155749/PyBERT.zip)

capn-freako, 2022-07-21:
Thanks for the alert, Dennis! I'll take a look.
-db

capn-freako, 2022-07-23:
Differences confirmed. Will investigate, this weekend.

capn-freako, 2022-09-26:
Hi Dennis,

Sorry about the delay on this. :(

It looks like this is fixed, as of the latest release (v3.5.7):

```
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT/misc/Support/DennisGhan/PyBERT (master)
$ ll
total 8
-rw-r--r-- 1 capnf 197609 1602 Jul 22 18:13 JESD204B_9.8304Gbps_Simberian_SL.yaml
-rw-r--r-- 1 capnf 197609 1600 Sep 25 17:44 JESD204B_9.8304Gbps_Simberian_SL_saved.yaml
(pybert-dev)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT/misc/Support/DennisGhan/PyBERT (master)
$ diff JESD204B_9.8304Gbps_Simberian_SL.yaml JESD204B_9.8304Gbps_Simberian_SL_saved.yaml
28c28
<   - 0
---
> - 0
```

Would you confirm, please, so I can close this one?

Thanks!
-db

dghan0219, 2022-09-30:
Configuration file loaded and then saved match.  Issue is closed.

closed 2022-09-30 by dghan0219