capn-freako/PyBERT #106: Attempt Pip Installation

capn-freako, 2022-10-20:
With the recent modernization of the [_Pip_ package installation infrastructure](https://pip.pypa.io/en/stable/reference/build-system/pyproject-toml/), it may be time to again attempt to get _PyBERT_ into a "_Pip_ installable" state.

Admittedly, our past attempts at this have been pretty miserable failures.
The general sense I have about this is that only real Python wizards succeed and not in a way that can be easily exported to the general _PyBERT_ user base, who must continue to use _Anaconda_ to get a reliable installation of _PyBERT_ up and running.
However, in briefly reviewing the advancements embodied by [PEP 517](https://peps.python.org/pep-0517/) and [PEP 518](https://peps.python.org/pep-0517/), it seems like it might be worth trying again.
And it would be well worth the effort if we succeed, because many folks, who want to use _PyBERT_, can't use _Anaconda_ for various reasons.

**Note:** It's important that we do this in a way that leaves us with a "dual-mode" approach.
That is, we want to end up in a state where users can install _PyBERT_ (and/or _PyAMI_), using either `conda` or `pip`, with little hassle.
(I think 30 minutes is a good "threshold" for the amount of time it should take to get up and running.)

I invite _anyone_ with experience at setting up Python packages for such "dual-mode" installation capabilities to comment here, regarding their experiences.
Also, if you are one of the gifted few, who've been successful at getting _PyBERT_ installed and running properly using `pip`, as opposed to `conda`, _please_ post your recipe here for others to try.

Thank you!
-db

jdpatt, 2022-10-21:
I should have the same three PRs for pybert which should allow pip usage again!  The main open/question is around all the various git submodules.  Are they just for convenience sake when building the conda packages or have you "vendored" them and made source code changes that their pip equivalent may not have.  If we can assume that they are vanilla copies, that makes things easy. :)

jdpatt, 2022-10-21:
Awesome, I assumed but thanks for the confirmation!

capn-freako, 2022-11-06:
Hi @jdpatt ,

I'm having trouble making sense of this.
It looks to me like `pyibis-ami` v3.5.7 is installed in the Tox `py38` test environment.
Why is it complaining that it can't find it?

Thanks,
-db

```
(pybert-dev)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (add-tests)
$ tox -e py38 -r
GLOB sdist-make: C:\Users\capnf\Documents\GitHub\PyBERT\setup.py
py38 recreate: C:\Users\capnf\Documents\GitHub\PyBERT\.tox\py38
py38 installcondadeps: pyibis-ami
py38 installdeps: pytest-cov, numpy, Cython
py38 inst: C:\Users\capnf\Documents\GitHub\PyBERT\.tox\.tmp\package\1\PyBERT-3.5.8.zip
ERROR: invocation failed (exit code 1), logfile: C:\Users\capnf\Documents\GitHub\PyBERT\.tox\py38\log\py38-3.log
================================== log start ==================================
Processing c:\users\capnf\documents\github\pybert\.tox\.tmp\package\1\pybert-3.5.8.zip
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
{Many successful package installations omitted.}
ERROR: Could not find a version that satisfies the requirement pyibis-ami>=3.5.7 (from pybert) (from versions: 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.9, 1.1.1, 1.1.2, 1.1.3, 1.1.4, 1.1.5, 1.1.6, 1.1.7, 1.1.8, 2.0.0, 2.0.1, 2.0.2, 2.0.3, 2.0.4, 2.0.5, 2.2.0, 2.2.1, 2.2.2)
ERROR: No matching distribution found for pyibis-ami>=3.5.7

=================================== log end ===================================
___________________________________ summary ___________________________________
ERROR:   py38: InvocationError for command 'C:\Users\capnf\Documents\GitHub\PyBERT\.tox\py38\python.EXE' -m pip install --exists-action w '.tox\.tmp\package\1\PyBERT-3.5.8.zip' (exited with code 1)

$ conda activate .tox/py38

$ conda list pyibis-ami
# packages in environment at C:\Users\capnf\Documents\GitHub\PyBERT\.tox\py38:
#
# Name                    Version                   Build  Channel
pyibis-ami                3.5.7                    py38_1    local
```

capn-freako, 2022-11-06:
It's getting curiouser and curiouser:

With my normal `pybert-dev` environment activated, _Pip_ "sees" the correct version:

```
(pybert-dev)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (add-tests)
$ pip show pyibis-ami
Name: PyIBIS-AMI
Version: 3.5.7
{snip}
Location: c:\users\capnf\documents\github\pybert\pyami
Requires: click, empy, matplotlib, numpy, parsec, scipy, traits, traitsui
Required-by: PyBERT
```

But when I activate _Tox_'s `py38` environment, it sees v3.4.1:

```
(pybert-dev)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (add-tests)
$ conda activate .tox/py38/

(C:\Users\capnf\Documents\GitHub\PyBERT\.tox\py38)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (add-tests)
$ pip show pyibis-ami
Name: PyIBIS-AMI
Version: 3.4.1
{snip}
Location: c:\users\capnf\documents\github\pybert\.tox\py38\lib\site-packages
Requires: click, empy, matplotlib, numpy, parsec, scipy, traits, traitsui
Required-by:
```

And `pip` thinks that `conda` put it there:

```
(C:\Users\capnf\Documents\GitHub\PyBERT\.tox\py38)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (add-tests)
$ pip list -v
Package         Version   Location                                                           Installer
--------------- --------- ------------------------------------------------------------------ ---------
attrs           22.1.0    c:\users\capnf\documents\github\pybert\.tox\py38\lib\site-packages pip
{snip}
PyIBIS-AMI      3.4.1     c:\users\capnf\documents\github\pybert\.tox\py38\lib\site-packages conda
```

But, `conda` doesn't seem to know about v3.4.1, from within either environment!:

```
(C:\Users\capnf\Documents\GitHub\PyBERT\.tox\py38)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (add-tests)
$ conda search --info --use-local -c dbanas -c defaults -c conda-forge pyibis-ami
Loading channels: ...working... done
pyibis-ami 3.5.6 py38_1
-----------------------
file name   : pyibis-ami-3.5.6-py38_1.tar.bz2
name        : pyibis-ami
version     : 3.5.6
{snip}
url         : file:///C:/Users/capnf/anaconda3/conda-bld/win-64/pyibis-ami-3.5.6-py38_1.tar.bz2
{snip}

pyibis-ami 3.5.7 py38_1
-----------------------
file name   : pyibis-ami-3.5.7-py38_1.tar.bz2
name        : pyibis-ami
version     : 3.5.7
{snip}
url         : file:///C:/Users/capnf/anaconda3/conda-bld/win-64/pyibis-ami-3.5.7-py38_1.tar.bz2
{snip}

pyibis-ami 3.5.7 py38_1
-----------------------
file name   : pyibis-ami-3.5.7-py38_1.tar.bz2
name        : pyibis-ami
version     : 3.5.7
{snip}
url         : https://conda.anaconda.org/dbanas/win-64/pyibis-ami-3.5.7-py38_1.tar.bz2

(C:\Users\capnf\Documents\GitHub\PyBERT\.tox\py38)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (add-tests)
$ conda deactivate

(pybert-dev)
capnf@DESKTOP-G84ND7C MINGW64 ~/Documents/GitHub/PyBERT (add-tests)
$ conda search --info --use-local -c dbanas -c defaults -c conda-forge pyibis-ami
Loading channels: ...working... done
{same results as above}
```

capn-freako, 2022-11-06:
I posted [this SO question](https://stackoverflow.com/q/74336732/1549736).

capn-freako, 2022-11-06:
Hi @jdpatt ,

I think I found the source of the mysterious "3.4.1" version number.
It was in your new `PyAMI/pyproject.toml` file.
However, changing it to "3.5.7" hasn't fixed my problem.
And I'm still finding version 3.4.1 in `.tox/py38/Lib/site-packages/`, despite having given the `-r` option to the `tox` command.

Might I need to rebuild something for _Tox_ after having changed the version number in `PyAMI/pyproject.toml`?

Thanks,
-db

capn-freako, 2022-11-07:
Okay, doing a `conda` rebuild after correcting the version number in `pyproject.toml` seems to have fixed the breakage.

capn-freako, 2022-11-07:
Hi @jdpatt ,

I've got a new problem: I keep bumping into this:

```
_______________ ERROR collecting tests/test_basic/test_perf.py ________________
test_basic\test_perf.py:4: in <module>
    dut = PyBERT(gui=False)
..\.tox\py38\lib\site-packages\pybert\pybert.py:658: in __init__
    my_run_simulation(self, initial_run=True)
..\.tox\py38\lib\site-packages\pybert\sim\simulation.py:282: in my_run_simulation
    temp.resize(len(t))
E   ValueError: cannot resize an array that references or is referenced
E   by another array in this way.
E   Use the np.resize function or refcheck=False
```

These `<NumPy ndarray>.resize()` callse have never caused me trouble before.
Do you know what it is about running under _Tox_ that is triggering this?

capn-freako, 2022-11-10:
> Ah, yeah. I did. Its not tox, just debugging which tries to inspect the array and fails. Do you remember my PR, (#99) a while back to change that but you found that ndarray.resize() is different from np.resize(ndarray) and how it pads the remaining array? That is exactly why I raised the issue. Sadly, I think I just avoided functions that used that when unit testing.

Yes, I do remember that.
I've been able to fix this, by adding `refcheck=False` to all those `resize()` calls.
I just hope I haven't exposed the code to mysterious bugs in the future. :(

capn-freako, 2022-12-08:
Hi @jdpatt ,

With our _NumPy_ performance problem solved and just 3 remaining issues in your _Milestones_ list, I'd like to push for a pre-Xmas release.
It'd be great to give people the Xmas/NewYear break to "play" w/ the new code and generate feedback.

You game?

Thanks,
-db

jdpatt, 2023-01-01:
PyBERT is available on pypi under the name `pipbert` since pybert is currently taken.  Install instructions have the latest name/directions.  https://github.com/capn-freako/PyBERT/wiki/instant_gratification

closed 2023-01-01 by jdpatt