capn-freako/PyBERT #111: rx_in is saved in the results file but has no plot to return too.

jdpatt, 2022-11-27:
When loading any saved results file, an exception is silently printed in the console saying that there is no `rx_in_ref` plot which there isn't.

```python
Exception occurred in traits notification handler for event object: TraitChangeEvent(object=<chaco.array_plot_data.ArrayPlotData object at 0x131737ae0>, name='data_changed', old=<undefined>, new={'added': ['rx_in_ref']})
Traceback (most recent call last):
  File "/git/PyBERT/.venv/lib/python3.9/site-packages/traits/observation/_trait_event_notifier.py", line 122, in __call__
    self.dispatcher(handler, event)
  File "/git/PyBERT/.venv/lib/python3.9/site-packages/traits/observation/observe.py", line 27, in dispatch_same
    handler(event)
  File "/git/PyBERT/.venv/lib/python3.9/site-packages/chaco/plot.py", line 1466, in _data_update_handler
    self._get_or_create_datasource(name)
  File "/git/PyBERT/.venv/lib/python3.9/site-packages/chaco/plot.py", line 1423, in _get_or_create_datasource
    raise ValueError(
ValueError: Unhandled array shape in creating new plot: ()
```

Looking back at the commit history, https://github.com/capn-freako/PyBERT/commit/0ef8d8f20cbd5f5666952cc5111c4f5b96e8fb51 adds rx_in to the items that should be loaded back into plots.  @capn-freako was the intent to just reload rx_in and should be a new attribute that is loaded back in differently than the plots or is there something that I'm missing?  Currently, rx_in is never reloaded since this exception is raised.

capn-freako, 2022-12-08:
Hi @jdpatt ,

Would you _assign_ these to me when the ball is in my court, please?
I only found this, because I was perusing the issues you placed in the _v4.0.0_ milestone you created.
Thanks for creating that milestone and adding the relevant issues to it, by the way!
I never knew about this feature, previously. :(
It's perfect for a new release. :)

Thanks,
-db

P.S. - Looking into this, now...

jdpatt, 2022-12-08:
Ah yeah, I should have done that.  Wilco!

capn-freako, 2022-12-08:
@jdpatt ,

After reviewing the code, I saw no reason not to simply remove `rx_in` from the list of saved items.
Let me know if you think that was a mistake.

Thanks,
-db

commit eca202c8 (2022-12-08): Removed `rx_in` from saved results items.

Fixes #111
commit 1eb02e88 (2022-12-11): Change ButtonHandler to MenuBar (#116)

* feat: make it where you can clear reference plots

it doesn't do anything if there are none.

* refactor: loading/saving tests should use fixtures

* fix: #111 by commenting out; not final solution

* doc: formatting only

* Change from global buttons to traditional menuBar

* rename button

* add id based off one example

* fix: removed the double newline printing for msgs

* rename help time to info

"help" exists in the menuBar now.

* revert: didn't intend to remove.
closed 2022-12-08 by capn-freako