capn-freako/PyBERT #112: feat: Keyboard Shortcuts

jdpatt, 2022-11-27:
Traitsui allows the use of keyboard shortcuts thru "accelerators".  Everything that I see says the ButtonHandler should support them as well.  Try it out and/or alternatively try a menuBar.  https://docs.enthought.com/envisage/tasks_user_manual/menus.html

- Run "Ctrl + R"
- Open Config "Ctrl + O"
- Save Config "Ctrl + S"
- Toggle Debug Mode "Ctrl + ~"

capn-freako, 2022-12-08:
Hi @jdpatt ,

I'm confused. Have you made this addition to the code and are asking someone to kick its tires, or are you suggesting that the above code snippet be added/tried?

Thanks,
-db

jdpatt, 2022-12-08:
That was copied from my develop branch as my starting point.  I'll open a PR tonight for this and clearing reference waveforms (#113 ) since it is closely related and done for you to play with.  I still need to try your other toolkits but with qt, all things work.

PR #116 (2022-12-11): Change ButtonHandler to MenuBar
This PR closes #112 and closes #113 by making two changes to the user interface.

### Adds
- It adds a button to be able to clear any waveforms that are loaded by the user instead of requiring them to close/re-open the GUI to clear them. 
- It adds keyboard short cuts so that you can run, open, save quickly in the GUI.  The full list of short cuts:
    - Run "Ctrl + R"
    - Quit "Ctrl + Q"
    - Open Config "Ctrl + O"
    - Save Config "Ctrl + S"
    - Save Config As "Ctrl + Shift + S"
    - Toggle Debug Mode "Ctrl + `"

### Changes
- The button handler at the  [... cut]
closed 2022-12-11 by capn-freako