capn-freako/PyBERT #131: Shortcut to start pybert

buenoshun, 2023-06-14:
Is it possible to create a windows desktop shortcut that starts anaconda with a doubleclick, that automatically issues the two commands that are needed to start pybert? I'm having hard time remembering commands to type in every time I want to start the pybert application. 

instead of this:
%windir%\System32\cmd.exe "/K" C:\Users\bueno\miniconda3\Scripts\activate.bat C:\Users\bueno\miniconda3
this opens the console. then type:
conda activate pybert
pybert

Just have one shortcut that does it all?
Regards,
Istvan Nagy

buenoshun, 2023-06-16:
> Hey Istvan,
>
> I can't try it out myself right now but combing these two links, but I think
> so!
>
> -
> https://stackoverflow.com/questions/9738434/run-a-command-prompt-command-from-desktop-shortcut
> -
> https://stackoverflow.com/questions/35982273/use-anaconda-environment-without-activate-e-g-in-crontab
>
> I think you were super close. Try one of the following when you create the
> shortcut:
>
> - `%windir%\System32\cmd.exe /k conda activate pybert & pybert`
> - `%windir%\System32\cmd.exe /k conda run -n pybert pybert`
>
> For others in the future that are using pip/virtualenv, you could just
> directly use the console entry point that gets created when you install it.
> For example on OS X, I can use "automator" and "run shell script" with just
> `/Users/david/git/PyBERT/.venv/bin/pybert`.

Hi,

The issue is miniconda does not start with "conda" on my computer.
It starts with:
%windir%\System32\cmd.exe "/K"
C:\Users\bueno\miniconda3\Scripts\activate.bat
C:\Users\bueno\miniconda3
I cannot append anything to that.

If I make a copy of that activate.bat and call it activate2.bat, edit
it, add "activate pybert" and "pybert" at the end, then change my
shortcut to refer to the new bat file, then I get one step closer, it
will activate pybert, but the last command "pybert" does not go
through. I still have to type that in. every time there is an
activate, the command line changes owner, and forgets everything else
that was in the batch file.

jdpatt, 2023-06-19:
Googling around seems like you have to use the `call` command to not loose some of the .bat environment stuff but even easier, conda also honors/creates the console entry point!

Just create a shortcut directly to the `pybert.exe` in your env folder.  For me this was, `C:\Users\David\anaconda3\envs\pybert\Scripts\pybert.exe`.   You can update the "Start In" field to be where ever your pybert files are so that logs and save files are saved there vs. System32.

jdpatt, 2023-07-10:
Hopefully this worked for you.  I'll make sure to add this to the getting started and then close this within the week assuming that it worked.

buenoshun, 2023-07-10:
Hi,

Thanks.
The direct link worked for me.
C:\Users\bueno\miniconda3\envs\pybert\Scripts\pybert.exe

Istvan Nagy

closed 2023-07-10 by jdpatt