capn-freako/PyBERT #138: Make LICENSE discoverable to GitHub as BSD-2

mgielda, 2023-11-21:
**Describe the desired new or improved feature.**

According to https://github.com/jpeddicord/askalono the license of PyBERT is close to BSD 2-clause (similarity score is 0.971). However GitHub does not seem to be able to detect it as such. Did some testing and this is due to 3 things:

1) the LICENSE header with underline
2) lack of numbering in the "clauses" and space between the clauses and the NO WARRANTY text
3) the non-standard "All rights reserved World Wide" line.

The first two issues are probably very non-controversial, i.e. pertain to formatting only. The third one however might legitimately make this project not BSD-2 licensed (or at least not obviously so). Which would be a shame; most corporate legal departments will not permit usage of software licenses not on the OSI-approved list. It's hard to explain to them that it's only slightly different and essentially the same, i.e. the burden of proof lies on the developer.

The question is, could we fix this? I am not a lawyer of course, but I don't believe the "All rights reserved" notice has any legal meaning whatsoever these days (https://en.wikipedia.org/wiki/All_rights_reserved - see "Obsolescence" which mentions it's no longer necessary to state the assertion of copyrights explicitly since 2000).

So summing up, I think that the license could be changed to be detectable by GH with no legal impact (other than making the library look safer to use).

**Expected behavior**

License detected as in: https://github.com/mgielda/PyBERT/blob/master/LICENSE

If this suggestion makes sense, I can make a PR.

capn-freako, 2023-11-21:
Hi Michael,

Yes, I'm okay with this.
But, my intent was that _PyBERT_ be licensed under the BSD 3-clause license.

-db

capn-freako, 2023-11-23:
Okay, I've updated the text in the `LICENSE` file to be an exact replica of the BSD-3 license.
And _GitHub_ is now recognizing it as such.

closed 2023-11-23 by capn-freako