capn-freako/PyBERT #149: Extension of exportRes2dInv

IsabellaLorenzen, 2024-04-04:
Hallo!

I am extending the exportRes2dInv function so that it exports more than one IP-window and assigns the right IP-delay/integration times. 

I am currently filtering my data, and after the export from the extension i will run it through Res2dInv. 

If succesful, and it seems like it will be, are you interested in receiving the extended function? If so, where and how can i upload it?

Best regards, 
Isabella

capn-freako, 2024-04-04:
Hi Isabella,

Thank you for getting involved with PyBERT!
Yes, I would love to have your contribution. :)
The best way to contribute is:

1. Create your own fork of _PyBERT_.
2. Make your code changes to _your_ fork.
3. Create a _Pull Request_ from your fork to mine.

Let me know if you need help with any of these steps.

And welcome to the PyBERT team!

Best regards,
-db

IsabellaLorenzen, 2024-04-05:
Thank you!
I am new to gitHub, is it possible that you could elaborate the steps?

capn-freako, 2024-04-05:
Sure. It’s probably best if we hop on a Zoom chat, so I can just share my screen with you and give you a demo.
I’m pretty wide open today; do you have a free time slot?
-db

IsabellaLorenzen, 2024-04-05:
I have a free time slot now?

capn-freako, 2024-04-05:
Okay, sending you a Zoom invite now...

capn-freako, 2024-04-05:
Here you go:
https://us04web.zoom.us/j/75631549297?pwd=zOW41oXvgb7MIe4PzH9FaYJjbdhbx1.1 <https://us04web.zoom.us/j/75631549297?pwd=zOW41oXvgb7MIe4PzH9FaYJjbdhbx1.1>

capn-freako, 2024-06-16:
Isabella was trying to target a different project.
Closing this issue as non-apropos.

closed 2024-06-16 by capn-freako