capn-freako/PyBERT #151: Does not compile for apple silicon

keixthb, 2024-05-22:
## Build Failure Using PIP on MacOS M3 Pro

The pip install  behavior of ``PyBERT`` fails on apple silicon. PyBERT depends on the ``enable`` package found [here](https://github.com/enthought/enable), and due to this [issue](https://github.com/enthought/enable/issues/1067), I do not expect that the pip install will work until this is resolved. 

**To Reproduce**

Using Mac OS Sonoma M3-Pro:

```console
brew install python@3.9
git clone --depth=10 https://github.com/capn-freako/PyBERT
cd PyBERT
python3.9 -m venv temp
pip install --upgrade pip
pip install enable
```

### Error message

<details><summary>Error message</summary>

<pre>

Building wheels for collected packages: enable
  Building wheel for enable (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for enable (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [390 lines of output]
      running bdist_wheel
      running build
      running build_ext
      /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-jra5vjcf/overlay/lib/python3.9/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-install-v460lrh7/enable_546a2205101445b09e584f9dffa888af/kiva/_cython_speedups.pyx
        tree = Parsing.p_module(s, pxd, full_module_name)
      Compiling kiva/_cython_speedups.pyx because it changed.
      [1/1] Cythonizing kiva/_cython_speedups.pyx
      building 'kiva._cython_speedups' extension
      creating build
      creating build/temp.macosx-14-arm64-cpython-39
      creating build/temp.macosx-14-arm64-cpython-39/kiva
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -Ikiva -I/private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-jra5vjcf/overlay/lib/python3.9/site-packages/numpy/core/include -I/Users/keithhbova/Desktop/PyBERT/temp/include -I/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c kiva/_cython_speedups.cpp -o build/temp.macosx-14-arm64-cpython-39/kiva/_cython_speedups.o
      In file included from kiva/_cython_speedups.cpp:1266:
      In file included from /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-jra5vjcf/overlay/lib/python3.9/site-packages/numpy/core/include/numpy/arrayobject.h:4:
      In file included from /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-jra5vjcf/overlay/lib/python3.9/site-packages/numpy/core/include/numpy/ndarrayobject.h:12:
      In file included from /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-jra5vjcf/overlay/lib/python3.9/site-packages/numpy/core/include/numpy/ndarraytypes.h:1969:
      /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-jra5vjcf/overlay/lib/python3.9/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: "Using deprecated NumPy API, disable it with "          "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-W#warnings]
      #warning "Using deprecated NumPy API, disable it with " \
       ^
      1 warning generated.
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -Ikiva -I/private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-jra5vjcf/overlay/lib/python3.9/site-packages/numpy/core/include -I/Users/keithhbova/Desktop/PyBERT/temp/include -I/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c kiva/_hit_test.cpp -o build/temp.macosx-14-arm64-cpython-39/kiva/_hit_test.o
      creating build/lib.macosx-14-arm64-cpython-39
      creating build/lib.macosx-14-arm64-cpython-39/kiva
      clang++ -bundle -undefined dynamic_lookup -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk build/temp.macosx-14-arm64-cpython-39/kiva/_cython_speedups.o build/temp.macosx-14-arm64-cpython-39/kiva/_hit_test.o -o build/lib.macosx-14-arm64-cpython-39/kiva/_cython_speedups.cpython-39-darwin.so
      /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-jra5vjcf/overlay/lib/python3.9/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-install-v460lrh7/enable_546a2205101445b09e584f9dffa888af/kiva/_marker_renderer.pxd
        tree = Parsing.p_module(s, pxd, full_module_name)

      Error compiling Cython file:
      ------------------------------------------------------------
      ...
      # Thanks for using Enthought open source!
      import cython
      import numpy as np
      from numpy cimport uint8_t

      cimport _marker_renderer
              ^
      ------------------------------------------------------------

      kiva/_marker_renderer.pyx:14:8: '_marker_renderer.pxd' not found
[... 447 lines cut ...]
sphinxcontrib-qthelp==1.0.7
sphinxcontrib-serializinghtml==1.1.10
tomli==2.0.1
tomlkit==0.12.5
tox==4.15.0
traits==6.4.3
traitsui==8.0.0
types-PyYAML==6.0.12.20240311
typing_extensions==4.11.0
tzdata==2024.1
urllib3==2.2.1
virtualenv==20.26.2
zipp==3.18.2

</pre></details>

**Results**

Code appears to work fine in rhel otherwise:

[image]

**Additional Information**

You may be able to resolve this issue with ``enable`` on apple silicon by using anaconda, but I have not tested it myself.

capn-freako, 2024-06-16:
Hi @keixthb ,

Would you, please, give the [latest installation instructions](https://github.com/capn-freako/PyBERT/wiki/instant_gratification) a try and let me know what happens?

Thanks!
-db

keixthb, 2024-06-16:
@capn-freako Using python3.12, the result is the same:

```console
pip install enable
```

falis with:

<details><summary>Error message</summary>

<pre>

 pip install enable
Collecting enable
  Using cached enable-6.0.0.tar.gz (2.9 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy (from enable)
  Downloading numpy-2.0.0-cp312-cp312-macosx_14_0_arm64.whl.metadata (60 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.9/60.9 kB 950.6 kB/s eta 0:00:00
Collecting pillow (from enable)
  Using cached pillow-10.3.0-cp312-cp312-macosx_11_0_arm64.whl.metadata (9.2 kB)
Collecting traits>=6.2.0 (from enable)
  Downloading traits-6.4.3-cp312-cp312-macosx_10_9_universal2.whl.metadata (5.0 kB)
Collecting traitsui (from enable)
  Using cached traitsui-8.0.0-py3-none-any.whl.metadata (6.8 kB)
Collecting pyface>=7.2.0 (from enable)
  Using cached pyface-8.0.0-py3-none-any.whl.metadata (7.7 kB)
Collecting fonttools (from enable)
  Using cached fonttools-4.53.0-cp312-cp312-macosx_11_0_arm64.whl.metadata (162 kB)
Using cached pyface-8.0.0-py3-none-any.whl (1.3 MB)
Downloading traits-6.4.3-cp312-cp312-macosx_10_9_universal2.whl (5.0 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.0/5.0 MB 8.8 MB/s eta 0:00:00
Using cached fonttools-4.53.0-cp312-cp312-macosx_11_0_arm64.whl (2.2 MB)
Downloading numpy-2.0.0-cp312-cp312-macosx_14_0_arm64.whl (5.0 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.0/5.0 MB 10.1 MB/s eta 0:00:00
Using cached pillow-10.3.0-cp312-cp312-macosx_11_0_arm64.whl (3.4 MB)
Using cached traitsui-8.0.0-py3-none-any.whl (1.5 MB)
Building wheels for collected packages: enable
  Building wheel for enable (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for enable (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [399 lines of output]
      running bdist_wheel
      running build
      running build_ext
      /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-install-hs1cgd4f/enable_363a329928cc4367ab1ff46e5d4774dd/kiva/_cython_speedups.pyx
        tree = Parsing.p_module(s, pxd, full_module_name)
      Compiling kiva/_cython_speedups.pyx because it changed.
      [1/1] Cythonizing kiva/_cython_speedups.pyx
      building 'kiva._cython_speedups' extension
      creating build
      creating build/temp.macosx-14.0-arm64-cpython-312
      creating build/temp.macosx-14.0-arm64-cpython-312/kiva
      clang -fno-strict-overflow -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -Ikiva -I/private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/numpy/core/include -I/Users/keithhbova/Desktop/temp/temp/include -I/opt/homebrew/opt/python@3.12/Frameworks/Python.framework/Versions/3.12/include/python3.12 -c kiva/_cython_speedups.cpp -o build/temp.macosx-14.0-arm64-cpython-312/kiva/_cython_speedups.o
      In file included from kiva/_cython_speedups.cpp:1266:
      In file included from /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/numpy/core/include/numpy/arrayobject.h:5:
      In file included from /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/numpy/core/include/numpy/ndarrayobject.h:12:
      In file included from /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/numpy/core/include/numpy/ndarraytypes.h:1929:
      /private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: "Using deprecated NumPy API, disable it with "          "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-W#warnings]
      #warning "Using deprecated NumPy API, disable it with " \
       ^
      kiva/_cython_speedups.cpp:24411:13: warning: code will never be executed [-Wunreachable-code]
                  goto bad;
                  ^~~~~~~~
      2 warnings generated.
      clang -fno-strict-overflow -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -Ikiva -I/private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/numpy/core/include -I/Users/keithhbova/Desktop/temp/temp/include -I/opt/homebrew/opt/python@3.12/Frameworks/Python.framework/Versions/3.12/include/python3.12 -c kiva/_hit_test.cpp -o build/temp.macosx-14.0-arm64-cpython-312/kiva/_hit_test.o
[... 355 lines cut ...]
          super().run_command(command)
        File "/private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
[... cut ...]
        File "/private/var/folders/g8/zyb92nd97sqf8x7py60g9fc00000gn/T/pip-build-env-mqi44fb_/overlay/lib/python3.12/site-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
          raise CompileError(None, pyx_file)
      Cython.Compiler.Errors.CompileError: kiva/_marker_renderer.pyx
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for enable
Failed to build enable
ERROR: Could not build wheels for enable, which is required to install pyproject.toml-based projects

</pre></details>

closed 2024-06-16 by keixthb

capn-freako, 2024-06-17:
Great! Thank you for your perseverance. :)