capn-freako/PyBERT #160: can not load ad9081_jrx.ibs file into config block

GeorgeSJLai, 2025-02-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows, Linux, or MacOS]
 - Python Version [e.g. 3.7.4]
 - PyBERT Version [e.g. 3.3.0]

**Additional context**
Add any other context about the problem here.

capn-freako, 2025-02-27:
Hi George,

Thanks for reporting this issue!
I'm investigating it now.

-db

capn-freako, 2025-03-08:
Bug found and fixed.

[image]

capn-freako, 2025-03-08:
Please, note that the fix involves a slight fix to one of PyBERT's dependencies: the Enthought _Traits_ package.
I have submitted [this Pull Request](https://github.com/enthought/traits/pull/1841) and am awaiting acceptance by that project's administrators.

capn-freako, 2025-03-08:
Useful links for catching up w/ the last week's activity:

- [Traits/UI Issue 2064](https://github.com/enthought/traitsui/issues/2064)
- [_Stack Overflow_ question](https://stackoverflow.com/questions/79484832/why-does-a-modal-call-of-configure-traits-crash-when-dynamically-added-traits)
- [PyAMI Pull Request 50](https://github.com/capn-freako/PyAMI/pull/50)

capn-freako, 2025-03-14:
Hi @GeorgeSJLai ,

I just posted what I think is a fix to PyPi.
Would you, please, try the following and let me know if things are working for you now?

1. Execute the following commands from a prompt with the correct Python virtual environment activated:

    - `pip install --upgrade pipbert`
    - `pybert`

2. Load that same ADI IBIS-AMI model into PyBERT and try to run the simulation.

Thanks!
-db

capn-freako, 2025-03-17:
Fixed in Rel. v7.1.0.

closed 2025-03-17 by capn-freako