capn-freako/PyBERT #163: ami get wave error

SmartHYH, 2025-04-18:
[data.zip](https://github.com/user-attachments/files/19810629/data.zip)
Dear Support Team,

I encountered a critical issue while using PyBERT with AMI models. When attempting to run a simulation using my provided TX and RX IBIS-AMI model files, along with a corresponding channel file, the simulation crashes during the GetWave stage.

Here are the details of the issue:

I enabled AMI and selected the GetWave flow.

Upon running the simulation, the program throws an access violation error.

The error log is as follows:

Traceback (most recent call last):
  File "D:\PICStudio\PhotoCAD\.python\win_x64\3.8.10\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\gui\view.py", line 50, in run
    my_run_sweeps(self.the_pybert)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\models\bert.py", line 103, in my_run_sweeps
    my_run_simulation(self)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\models\bert.py", line 264, in my_run_simulation
    tx_out, _ = tx_model.getWave(self.x)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pyibisami\ami\model.py", line 361, in getWave
    self._amiGetWave(
OSError: exception: access violation writing 0x0000000000000000

I suspect that the issue may be caused by either:

An incompatibility or bug within the AMI DLL models,

Or invalid or uninitialized data passed to the model (e.g., waveform or configuration).

Could you please help identify the root cause or let me know how to further debug this issue?

I would be happy to provide the TX, RX, and channel files used in the simulation upon request.

Best regards,

SmartHYH, 2025-04-18:
[image]

SmartHYH, 2025-04-18:
[image]

capn-freako, 2025-04-19:
Please try upgrading to the latest version of _PyBERT_: `pip install --upgrade pipbert`.
If the error persists then please do a run in _debug_ mode, by checking the `Debug` option in the first configuration tab, and posting the console log here.

Thanks!
-db

SmartHYH, 2025-04-22:
[2025-04-22 10:42:04.609320]: PyBERT: Parsing Tx IBIS file, 'D:/RD_data/spice/ads/CHANNEL_SIM/data/TXPCIE/TXPCIE.ibs'...

[2025-04-22 10:42:04.610339]: IBISModel: pyibisami.ibis_file.IBISModel initializing in debug mode...

Parsed nodes:
 [('ibis_ver', 7.1), ('file_name', 'TXPCIE.ibs'), ('file_rev', '1.0'), ('component', {'TXPCIE': <pyibisami.ibis.model.Component object at 0x000002B18DABC7C0>}), ('model', {'TXPCIE': <pyibisami.ibis.model.Model object at 0x000002B194FA7DB0>})]

[2025-04-22 10:42:04.623430]: IBISModel: IBIS parsing errors/warnings:
Success!

[2025-04-22 10:42:04.624424]: IBISModel: There was an [Algorithmic Model] keyword in this model.
 If you wish to use the AMI model associated with this IBIS model,
 please, go the 'Equalization' tab and enable it now.

[2025-04-22 10:42:05.931019]: IBISModel: Done.

[2025-04-22 10:42:05.932020]: PyBERT: Result:
Success!
[2025-04-22 10:42:05.936162]: PyBERT: Parsing Tx AMI file, 'D:/RD_data/spice/ads/CHANNEL_SIM/data/TXPCIE\TXPCIE.ami'...
[2025-04-22 10:42:05.939426]: PyBERT: Success.
[2025-04-22 10:42:40.982162]: PyBERT: Parsing Rx IBIS file, 'D:/RD_data/spice/ads/CHANNEL_SIM/data/RXPCIE/RXPCIE.ibs'...

[2025-04-22 10:42:40.983161]: IBISModel: pyibisami.ibis_file.IBISModel initializing in debug mode...

Parsed nodes:
 [('ibis_ver', 7.1), ('file_name', 'RXPCIE.ibs'), ('file_rev', '1.0'), ('component', {'RXPCIE': <pyibisami.ibis.model.Component object at 0x000002B19664B6D0>}), ('model', {'RXPCIE': <pyibisami.ibis.model.Model object at 0x000002B19667ED10>})]

[2025-04-22 10:42:40.996396]: IBISModel: IBIS parsing errors/warnings:
Success!

[2025-04-22 10:42:40.997396]: IBISModel: There was an [Algorithmic Model] keyword in this model.
 If you wish to use the AMI model associated with this IBIS model,
 please, go the 'Equalization' tab and enable it now.

[2025-04-22 10:42:47.386676]: IBISModel: Done.

[2025-04-22 10:42:47.386676]: PyBERT: Result:
Success!
[2025-04-22 10:42:47.396691]: PyBERT: Parsing Rx AMI file, 'D:/RD_data/spice/ads/CHANNEL_SIM/data/RXPCIE\RXPCIE.ami'...
[2025-04-22 10:44:03.109422]: PyBERT: RL: 99.99999989999998, Cp: 0.0
[2025-04-22 10:44:06.261071]: PyBERT: Channel calculation time: 3.370422400000024
[2025-04-22 10:44:06.261071]: PyBERT: Channel convolution time: 0.024290500000006432
[2025-04-22 10:44:06.268519]: PyBERT: Tx IBIS-AMI model initialization results:
**Input parameters: (TXPCIE (EnableEQ ['False', 'True'])(Preshoot 0.0)(Deemphasis 0.0)(TapInterval 1.0))**
Output parameters: 
**Message: Failed to read AMI parameter TXPCIE.EnableEQ.**
Exception in thread Thread-1:
Traceback (most recent call last):
  File "D:\PICStudio\PhotoCAD\.python\win_x64\3.8.10\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\gui\view.py", line 50, in run
    my_run_sweeps(self.the_pybert)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\models\bert.py", line 103, in my_run_sweeps
    my_run_simulation(self)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\models\bert.py", line 258, in my_run_simulation
    tx_s = getwave_step_resp(tx_model)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\utility.py", line 1242, in getwave_step_resp
    tx_s, _ = ami_model.getWave(tmp)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pyibisami\ami\model.py", line 361, in getWave
    self._amiGetWave(
OSError: exception: access violation writing 0x0000000000000000
**Some parameters in the new version may not be recognized? I use the file generated by KeySight ADS 'ami builder.**

> Please try upgrading to the latest version of _PyBERT_: `pip install --upgrade pipbert`. If the error persists then please do a run in _debug_ mode, by checking the `Debug` option in the first configuration tab, and post the console log here.
> 
> Thanks! -db

> 请尝试升级到最新版本的 _PyBERT_： 。如果错误仍然存在，请通过检查第一个配置选项卡中的选项，在_调试_模式下运行，并在此处发布控制台日志。`pip install --upgrade pipbert``Debug`
> 
> 谢谢！ -分贝

SmartHYH, 2025-04-22:
**Some parameters in the new version may not be recognized? I use the file generated by KeySight ADS 'ami builder.**

[2025-04-22 10:42:04.609320]: PyBERT: Parsing Tx IBIS file, 'D:/RD_data/spice/ads/CHANNEL_SIM/data/TXPCIE/TXPCIE.ibs'...

[2025-04-22 10:42:04.610339]: IBISModel: pyibisami.ibis_file.IBISModel initializing in debug mode...

Parsed nodes:
 [('ibis_ver', 7.1), ('file_name', 'TXPCIE.ibs'), ('file_rev', '1.0'), ('component', {'TXPCIE': <pyibisami.ibis.model.Component object at 0x000002B18DABC7C0>}), ('model', {'TXPCIE': <pyibisami.ibis.model.Model object at 0x000002B194FA7DB0>})]

[2025-04-22 10:42:04.623430]: IBISModel: IBIS parsing errors/warnings:
Success!

[2025-04-22 10:42:04.624424]: IBISModel: There was an [Algorithmic Model] keyword in this model.
 If you wish to use the AMI model associated with this IBIS model,
 please, go the 'Equalization' tab and enable it now.

[2025-04-22 10:42:05.931019]: IBISModel: Done.

[2025-04-22 10:42:05.932020]: PyBERT: Result:
Success!
[2025-04-22 10:42:05.936162]: PyBERT: Parsing Tx AMI file, 'D:/RD_data/spice/ads/CHANNEL_SIM/data/TXPCIE\TXPCIE.ami'...
[2025-04-22 10:42:05.939426]: PyBERT: Success.
[2025-04-22 10:42:40.982162]: PyBERT: Parsing Rx IBIS file, 'D:/RD_data/spice/ads/CHANNEL_SIM/data/RXPCIE/RXPCIE.ibs'...

[2025-04-22 10:42:40.983161]: IBISModel: pyibisami.ibis_file.IBISModel initializing in debug mode...

Parsed nodes:
 [('ibis_ver', 7.1), ('file_name', 'RXPCIE.ibs'), ('file_rev', '1.0'), ('component', {'RXPCIE': <pyibisami.ibis.model.Component object at 0x000002B19664B6D0>}), ('model', {'RXPCIE': <pyibisami.ibis.model.Model object at 0x000002B19667ED10>})]

[2025-04-22 10:42:40.996396]: IBISModel: IBIS parsing errors/warnings:
Success!

[2025-04-22 10:42:40.997396]: IBISModel: There was an [Algorithmic Model] keyword in this model.
 If you wish to use the AMI model associated with this IBIS model,
 please, go the 'Equalization' tab and enable it now.

[2025-04-22 10:42:47.386676]: IBISModel: Done.

[2025-04-22 10:42:47.386676]: PyBERT: Result:
Success!
[2025-04-22 10:42:47.396691]: PyBERT: Parsing Rx AMI file, 'D:/RD_data/spice/ads/CHANNEL_SIM/data/RXPCIE\RXPCIE.ami'...
[2025-04-22 10:44:03.109422]: PyBERT: RL: 99.99999989999998, Cp: 0.0
[2025-04-22 10:44:06.261071]: PyBERT: Channel calculation time: 3.370422400000024
[2025-04-22 10:44:06.261071]: PyBERT: Channel convolution time: 0.024290500000006432
[2025-04-22 10:44:06.268519]: PyBERT: Tx IBIS-AMI model initialization results:
Input parameters: (TXPCIE (EnableEQ ['False', 'True'])(Preshoot 0.0)(Deemphasis 0.0)(TapInterval 1.0))
Output parameters: 
Message: Failed to read AMI parameter TXPCIE.EnableEQ.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "D:\PICStudio\PhotoCAD\.python\win_x64\3.8.10\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\gui\view.py", line 50, in run
    my_run_sweeps(self.the_pybert)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\models\bert.py", line 103, in my_run_sweeps
    my_run_simulation(self)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\models\bert.py", line 258, in my_run_simulation
    tx_s = getwave_step_resp(tx_model)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pybert\utility.py", line 1242, in getwave_step_resp
    tx_s, _ = ami_model.getWave(tmp)
  File "D:\WORKSPACE\BASE\.venv_BASE\lib\site-packages\pyibisami\ami\model.py", line 361, in getWave
    self._amiGetWave(
OSError: exception: access violation writing 0x0000000000000000

capn-freako, 2025-04-26:
Bug confirmed in PyBERT v7.2.2 under Windows 10 64-bit.

The problem is here:

```shell
ami_params_in = b"(Ethernet_Tx (EnableEQ ['False', 'True'])(FFE_Taps {0.0, 0.0, 0.0, 1.0, 0.0})(TapInterval 1.0)
```

The `EnableEQ` parameter is being sent the _list_ of possible values, instead of the user selected value from that list.

capn-freako, 2025-04-26:
Minimal test case:

```python
# Minimal test case for PyBERT Issue 163.

from pyibisami.ami.parser import AMIParamConfigurator

AMI_FILENAME = "Ethernet_Tx/Ethernet_Tx.ami"

with open(AMI_FILENAME, "rt") as ami_file:
    ami_contents = ami_file.read()

ami_cfg = AMIParamConfigurator(ami_contents)
print(ami_cfg.input_ami_params)
```

yields:

```shell
$ python test.py
{'root_name': 'Ethernet_Tx', 'EnableEQ': ['False', 'True'], 'FFE_Taps': '{0.0, 0.0, 0.0, 1.0, 0.0}', 'TapInterval': 1.0}
```

The `'EnableEQ': ['False', 'True']` is incorrect.
It should be either: `'EnableEQ': 'False'`, or: `'EnableEQ': 'True'`.

capn-freako, 2025-04-26:
Ah, the model is failing to read the `FFE_Taps` input parameter upon initialization.

Using this new test case:

```python
# IBIS-AMI model exerciser for PyBERT Issue 163.

from pyibisami.ami.model import AMIModel, AMIModelInitializer
from pyibisami.ami.parser import AMIParamConfigurator

AMI_FILENAME = "Ethernet_Tx/Ethernet_Tx.ami"
DLL_FILENAME = "Ethernet_Tx/Ethernet_Tx_x64.dll"

with open(AMI_FILENAME, "rt") as ami_file:
    ami_contents = ami_file.read()

ami_cfg = AMIParamConfigurator(ami_contents)
ami_cfg()
print(ami_cfg.input_ami_params)

ami_init = AMIModelInitializer(ami_cfg.input_ami_params)
ami_model = AMIModel(DLL_FILENAME)
ami_model.initialize(ami_init)
print(ami_model.msg)
```

yields:

```shell
$ python test2.py
{'root_name': 'Ethernet_Tx', 'EnableEQ': True, 'FFE_Taps': '{0.0, 0.0, 0.0, 1.0, 0.0}', 'TapInterval': 1.0}
b'Failed to read AMI parameter Ethernet_Tx.FFE_Taps. '
```

capn-freako, 2025-04-26:
None of these work:

```python
    params_in = {
        'root_name': 'Ethernet_Tx',
        'EnableEQ': False,
        # 'FFE_Taps': '{0.0, 0.0, 0.0, 1.0, 0.0}',
        # 'FFE_Taps': [0.0, 0.0, 0.0, 1.0, 0.0],
        # 'Ethernet_Tx.FFE_Taps': '{0.0, 0.0, 0.0, 1.0, 0.0}',
        # 'FFE_Taps': '{0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0}',
        'FFE_Taps': '0.0, 0.0, 0.0, 1.0, 0.0',
        'TapInterval': 1.0
    }
```

capn-freako, 2025-04-29:
Okay, @SmartHYH , this should be fixed.
Please, try:

`pip install --upgrade PyIBIS-AMI`

And then re-run PyBERT, using that same model.
Let me know if it's not working for any reason.

Thanks,
-db

commit 472a3532 (2025-04-30): Added explicit double-quotes around AMI parameters of type: String.

Fixes #163
PR #164 (2025-04-30): Added explicit double-quotes around AMI parameters of type: String.
Fixes #163
commit cacc4360 (2025-06-08): Added explicit double-quotes around AMI parameters of type: String. (#164)

Fixes #163