capn-freako/PyBERT #168: Add a Viterbi Decoder to the Native PyBERT Rx Model.

capn-freako, 2025-06-10:
**Describe the desired new or improved feature.**
Add a _Viterbi_ decoder to the PyBERT native Rx model, so that _Maximum Likelihood Sequence Detection_ (MLSD) may be modeled.

**Additional context**
Make sure to include GUI elements needed to explain MLSD and the Viterbi implementation of it.

closed 2025-06-22 by capn-freako

capn-freako, 2025-06-24:
Good catch, @jdpatt !

I just pushed a fix to `master`.
Can you pull it and test things on your end, before I upload v8.1.1 to PyPi?

Thanks!
-db

jdpatt, 2025-06-25:
I don't get the same error and my simulation completes with all native items.  

FWIW, Perf does drop below 1Mbps/min for me which would cause a test case to fail if we were add it to the various workflows tested.  We don't do that today but assuming this should be considered.

capn-freako, 2025-06-26:
Thanks, @jdpatt !
Yeah, Viterbi is a real performance killer. :(
Feel like digging in and speeding it up?

Looking forward to Raleigh,
-db